krzysztofkuczek/QCZEK_LRS #1: switch to minimal RF (TX part) power when SBUS input is in use and there is no SBUS signal

krzysztofkuczek, 2019-09-03:
proposed by RocketMouse
"I would like to propose to make different logic here for no SBUS input. If there is no SBUS-in then it's better to switch TX off. It will be useful for a relay in case radio is switched off, let's say for some flight preparation, and QLRS will also stay off until radio will be switched on again. It will help to keep module cold and save relay powering battery."

MFlghts, 2019-09-05:
Hi Kris, 
Is there any reason to still keep it at lowest power while there is no input, or it will be alright to switch it off completely?

krzysztofkuczek, 2019-09-05:
Yes it its. Even without correct sbus signal you can transmit raw telemetry stream from ground station to uav....

MFlghts, 2019-09-05:
Nice! I didn't know about such a feature. Can you please explain how can it be used in real flight scenario?

krzysztofkuczek, 2019-09-06:
I was thinking about it and it will be set to minimum power level set by the user when no correct cppm/pxx/sbus signal comes.
Regarding your question, it ca be use for mission control, without radio... i thinking about extending LTM protocol to make it possible.

krzysztofkuczek, 2019-09-08:
done

closed 2019-09-08 by krzysztofkuczek

MFlghts, 2019-09-08:
Great! Thank you!