krzysztofkuczek/QCZEK_LRS #6: no of satelites not displayed with UBLOX temeletry

krzysztofkuczek, 2019-09-03:


krzysztofkuczek, 2019-09-05:
needs to be confirmed if user enabled ublox msg MSG_SOL

giacomo892, 2019-09-06:
Can you just make the UBLOX telemetry work only if NAV_PVT message is enabled?

krzysztofkuczek, 2019-09-06:
but why ? :)

giacomo892, 2019-09-06:
Simpler approach. Just one message to decode to have all the data you need.
Modern FC (at 10Hz) just enable NAV_PVT on the module, so if you want to tap the GPS TX wire to connect it to QLRS you are already out of luck

krzysztofkuczek, 2019-09-06:
well, well, but this message is 90 bytes long, and i don't have enough ram for processing.
this change requires testing, to make sure that heap will not be overflowed ...
will do that, but not now.
sorry

giacomo892, 2019-09-06:
don't worry, my was just an improvement suggestion :)