krzysztofkuczek/QCZEK_LRS #7: [Feature request] Send RSSI in dbm and SNR in db via S.Port telemetry, not raw register values

DiscoMan18, 2019-09-03:
As per your request from RCG thread: Also would appreciate a lot if all the math for converting reg values to a correct RSSI in dbm and SNR in db would be done in firmware instead of putting out raw values of corresponding registers to S.Port telemetry feed. They anyway need to be converted on user's end otherwise they are useless.

DiscoMan18, 2019-09-03:
Also would like to mention that it would be better to output SNR values with some kind of simple offset instead of using two's complement to represent negative numbers. For example, the value of 100 should be considered as 0, 101 as +1 db, 99 as -1 db, etc. This way it will be much easier to display SNR including its negative values on Taranis, by just adding this predefined offset in the settings of corresponding telemetry sensor on Taranis.

krzysztofkuczek, 2019-09-04:
hi @DiscoMan18 
I just make small experiment with different powers and attenuator.
Could you please check the results, and proposed smart telemetry calculation.
thx
[rssi_snr.xlsx](https://github.com/krzysztofkuczek/QCZEK_LRS/files/3576118/rssi_snr.xlsx)

DiscoMan18, 2019-09-07:
You calculations look correct. From the spreadsheet I can see that you used 32 as an offset for SNR values. The only question arises is this one: why did you use -157 as an offset for RSSI values? Shouldn't we use -164? Or -164 goes for 433 mhz modules and -157 for 868 mhz band?

krzysztofkuczek, 2019-09-08:
hi @DiscoMan18 
I made correction to snr and rssi calculation, then intalled 70dB attenator on rx o my desk and want away with my radio .... and now it looks like following for 1W (868MHz) power (sometimes switched to lower, but i cause link lost)

[image]

I seems to be that -112 dBm is minimum for two directional link working... in such a condition...
and snr is 6, but maybe there are low level of RF noise... i don't really know.
Anyway now i send raw RSSI reg to sport and there is need to substract 164 or 157 to display correct dBm depending on freqency
The SNR is calculated like following, where SX1278_SNR_OFFSET is 64 so you need to subtract 64 on taranis side...

```
int16_t SX1278ReadPackageSNR(void)
{
    int16_t snr;
    int8_t snrRegVal;
    snrRegVal = SX1278ReadReg(LR_RegPktSnrValue);
    if ( snrRegVal >= 0 )
    {
       snrRegVal+=2; 
    }else
    {
       snrRegVal-=2;  
    }
    snrRegVal /= 4;
    snr = snrRegVal + SX1278_SNR_OFFSET;
    if (snr < 0)
    {
        snr = 0;
    }
    return snr;
}
```

what do you think?
Kris

DiscoMan18, 2019-09-08:
> Anyway now i send raw RSSI reg to sport and there is need to substract 164 or 157 to display correct dBm depending on freqency

Let's make it clear finally)) -164 is and offset for 433 and -157 for 868, or viceversa?

`if ( snrRegVal >= 0 )
    {
       snrRegVal+=2; 
    }else
    {
       snrRegVal-=2;  
    }`

I didn't understand this part of your code... What's this for? Shouldn't be here a conversion from two's complement value, read from register? Or it's done inside of SX1278ReadReg() ?

krzysztofkuczek, 2019-09-08:
Hi,
-164 is for 433 and -157 is for 868, 915 and higher freq.
Adding or subtracting 2 is for rounding to nearest integer...

krzysztofkuczek, 2019-09-08:
done

closed 2019-09-08 by krzysztofkuczek

DiscoMan18, 2019-09-08:
Krzystof, and what about the conversion from two's complement internal representation of snr? Where do you do this? Inside of SX1278ReadReg() ?