krzysztofkuczek/QCZEK_LRS #9: Mavlink angle/heading

urlu75, 2019-09-03:
The angle seems to be divided by 100 when sent through Mavlink.
[RCG link](https://www.rcgroups.com/forums/showpost.php?p=41007283&postcount=2683)

here is the result with a transparent serial link :
mavlink_msg_global_position_int_get_hdg : 11914
mavlink_msg_vfr_hud_get_heading : 11900

here is the the result with QLRS :
mavlink_msg_global_position_int_get_hdg : 116
mavlink_msg_vfr_hud_get_heading : 116

Heading with the serial transparent link is 100 times bigger .

krzysztofkuczek, 2019-09-04:
hi @urlu75 ,
How did you get such a results?
I think that we have problem only with global possition.
Take a look at ardu plane raw data...
[image]

Kris

urlu75, 2019-09-04:
wierd.
These values come from the debug of zc6buj antenna tracker.
First with a serial link and then with QLRS.

Below is the code that has been change:
```
          ap_vz = mavlink_msg_global_position_int_get_vz(&msg);               // Ground Z Speed (Altitude, positive down), expressed as m/s * 100
          ap_hdg = mavlink_msg_global_position_int_get_hdg(&msg);             // Vehicle heading (yaw angle) in degrees * 100, 0.0..359.99 degrees          ap_ap_amsl = mavlink_msg_attitude_get_yaw(&msg);                // Yaw angle (rad, -pi..+pi)
          #ifdef QLRS
            ap_hdg = ap_hdg * 100;  //  Compensate for QLRS heading already divided by 100
          #endif
          if (!gpsGood) {
            gpsGood = true;
            #ifdef Debug_Status
              Debug.println("gpsGood=true");  
            #endif
          }
          new_GPS_data = true;

          cur.lat =  (float)ap_lat / 1E7;
          cur.lon = (float)ap_lon / 1E7;
          cur.alt = ap_amsl24 / 1E3;
          cur.hdg = ap_hdg / 100; 
```

I'm gonna double check asap.

krzysztofkuczek, 2019-09-05:
ok :)

krzysztofkuczek, 2019-09-05:
Imlemeted.

closed 2019-09-05 by krzysztofkuczek