krzysztofkuczek/QCZEK_LRS #13: SBUS failsafe. Add on option to keep channels as they are

giacomo892, 2019-09-06:
When there is a failsafe and SBUS is used, add an option to keep all the channels as they are and just set failsafe flag.
Will simplify a lot the setup without any risk of misconfiguration

krzysztofkuczek, 2019-09-29:
hi @giacomo892 
but what FC will read sbus frame when failsafe flag is set?

giacomo892, 2019-09-29:
No one of the modern ones at least since some months. 

Only use imho is to set a pan/tilt to a certain position during a fail safe event, but I doubt you can do that since when FS flag is set, FC ignores the updates even for the pan/tilt channels.

We can only keep M0/M1 pwm output failsafe values in the case someone decides to connect something to them and drop the rest saving RAM, FLASH and time since the configuration is kinda pointless.

What do you think now?