krzysztofkuczek/QCZEK_LRS #14: [new project] QLRS as pure telemetry link

giacomo892, 2019-09-06:
Giving as granted your valuable experience with the modules used by QLRS would be great to start a new project. 

Drop RX input / output and make QLRS telemetry link.

Allowing to send data transparently or even better having the option to allign MSP or MAVLINK packets in order to minimise packet losses and improve efficiency.
Would be great to be allowed to change coding and spreading settings too 

This would outperform the original firmware, which cannot handle MSP and MAVLINK and wipe out of the market SIK radios

krzysztofkuczek, 2019-09-09:
hi @giacomo892 
Good idea, do you have any experiance with SIK radio modules, what is minimal transfer which make possible planning mission, etc...
Kris

giacomo892, 2019-09-09:
Hi Kris, we can take a look around as soon we have time.
Sik radios should be well documented.

On the other hand the as we sync on the messages and make sure we don't send out truncated packets, it should work just fine even if we have to drop some packets.

giacomo892, 2019-09-09:
@krzysztofkuczek Seems like by default the radios have an air speed of 64kbps.
This rate transports 32kbps of real data (because by default half of it is used by ECC and FEC)

If you want to have better range you need to drop the rate on SIK radios too and rate and I think it can work quite well with a quarter of that speed. It is still a lot of data if you think about it... Can we get like 8Kbps (of payload) with our modules?

mohaoahmadi, 2019-10-11:
@krzysztofkuczek with the current Lora I think it’ll be a bit tricky as it doesn’t support higher bandwidth. The good news is, Semtec are making LoRa for the 2.4G frequency. Please have a look at the link https://www.semtech.com/products/wireless-rf/24-ghz-transceivers

Good luck :)

RMFPV, 2020-06-27:
Some useful links
https://ardupilot.org/copter/docs/common-3dr-radio-advanced-configuration-and-technical-information.html#common-3dr-radio-advanced-configuration-and-technical-information

https://github.com/ArduPilot/SiK