krzysztofkuczek/QCZEK_LRS #21: qczek as radar

artemsh, 2019-10-01:
Hi, Is it possible to use qczek lrs to transmit location data (lat, lon, altitude, heading, speed)(take data from mavlink telemetry) and receive data from other qczek lrs, like https://github.com/mistyk/inavradar-ESP32, in normal mode?
Data can be transferred to FC (INAV), for example, through the MSP2 protocol, the MSP2_COMMON_SET_RADAR_POS packet (https://github.com/iNavFlight/inav/blob/af08641f9539dbd4275a510accec977c431d4e6b/src/main/fc/fc_msp.c#L2411)

krzysztofkuczek, 2019-11-26:
well, also thinking about it, but it would required another module to recive/scan others transmisions....
Do you know if arduplane based fc (and osd) support radar?
Kris

artemsh, 2019-11-26:
I would not want to install another module. As an option, I think to use modules based on esp32, which will intercept telemetry and transmit to other modules via wi-fi. But here we need two-way telemetry. As far as I remember, in the lore module there is an opportunity to transfer 4 bytes with each packet. Will it be implemented in the future? And for example, supplement the LTM protocol for transmitting coordinates. Arduplan has not used it yet.

RMFPV, 2020-05-12:
Hi Kris, as I was told by another user there was an idea yo use ADSB to send second plane position.
Since I'm also very interested in this feature, I found there is already OSD support for this is almost implemented.
https://www.rcgroups.com/forums/showpost.php?p=44496075&postcount=125

I just didn't get how ADSB will be received - does it require second module, or it can be implemented in same RX as a telemetry out?