krzysztofkuczek/QCZEK_LRS #28: Taranis X7: Telemetry lost / restored. Telemetry constantly falls off / on. Switch flight modes (CPPM + S.Port)

AlexeyII, 2020-03-13:
There is also an audible warning: the engines disarmed / engines armed. Sounds are erroneous and do not affect the result. Also the mode switching in LUA to Taranis is not displayed correctly.

AlexeyII, 2020-03-13:
Firmware:
T433A_P_CPPM_P_PXX_T_SPORT_V1.92.s19
R433A_T_MAVLINK_P_CPPM_P_SBUS_V1.92.s19

AlexeyII, 2020-03-13:
For the test to be correct, I turned off the internal receiver(physically), there were no interference with the equipment.

shaman7036, 2020-07-03:
A similar problem! in version 2.0 telemetry does not work normally!

shaman7036, 2020-07-03:
A similar problem! in version 2.0 telemetry does not work normally! Moreover, in version 1.92 more or less something works! but also not without problems

krzysztofkuczek, 2020-09-29:
Fixed in 2.10

closed 2020-09-29 by krzysztofkuczek