krzysztofkuczek/QCZEK_LRS #29: SBAS + S.Port Telemetry. Тelemetry lost, LED on the transmitter goes out

AlexeyII, 2020-03-13:
SBAS + S.Port Telemetry. The module is connected to Taranis X7 and is binded with a receiver. Тelemetry lost when i moved stiks and the LED on the transmitter goes out at the end, the connection will be completely lost.

It is restored only if the Taranis X7 is rebooted. But it is lost again, after a sticks using.

AlexeyII, 2020-03-13:
There will be no complete disappearance of the control and the LED will be on when using CPPM + S.Port

AlexeyII, 2020-03-13:
Firmware:
T433A_P_CPPM_P_PXX_T_SPORT_V1.92.s19
R433A_T_MAVLINK_P_CPPM_P_SBUS_V1.92.s19

AlexeyII, 2020-03-13:
For the test to be correct, I turned off the internal receiver(physically), there were no interference with the equipment.

krzysztofkuczek, 2020-09-29:
Fixed in 2.10

closed 2020-09-29 by krzysztofkuczek