krzysztofkuczek/QCZEK_LRS #32: Mavlink broken on TX side, if sbus f/w.

whoim2, 2020-11-05:
Im try: 1.92, 2.1 f/w.
I need mavlink from air to ground.
Its work if TX f/w not sbus (cppm/frsky).
On sbus TX pin ground module send zerows.
e32-868t20s air and ground, im try f/w for 20s and 30s, work identical.

whoim2, 2020-11-05:
Additional problem (868t20s): cyclic lost connection if:
1) racemode on
2) high data rate on

whoim2, 2020-11-05:
New info: mavlink work! But baud 4800 on ground, ignored baud settings in configurator.
But many BT modules and all software for mavlink/android not work at 4800.

whoim2, 2020-11-07:
New additional problem: configurator 2.10 not save M0/M1 channels for pwm. Set channels, write to eeprom. disconnect/connect - see 0/0 in fields. 868T20S.

RomanLut, 2020-12-08:
Inav mavlink output (4800) and "Mavlink Encapsulation + SPORT" on ground works fine for me in 2.10. Sbus firmware. Baud settings are ignored indeed,  4800 are used. BT module should be configured for 4800.

whoim2, 2020-12-09:
> Inav mavlink output (4800) and "Mavlink Encapsulation + SPORT" on ground works fine for me in 2.10. Sbus firmware. Baud settings are ignored indeed, 4800 are used. BT module should be configured for 4800.

Yes, mavlink on ground only 4800. My bt minimum 9600 xD