krzysztofkuczek/QCZEK_LRS #33: proper conversion of flight mode from mavlink to sport for inav

RomanLut, 2020-12-08:
Flight mode is displayed incorrectly in the setup inav + qczek 2.1 (MAVLINK+SPORT) + inav lua telementry script.

With SPORT telemetry, lua script expects flight mode in Tmp1 sensor as described here:
https://github.com/iNavFlight/inav/b...s/Telemetry.md

When inav outputs mavlink, it encodes flight mode in baseFlags, systemState and mainly customMode flags of Hearbeat message:
https://github.com/iNavFlight/inav/b...etry/mavlink.c

so in order to display flight mode correctly, qczek TX module has to map mavlink heartbeat to Tmp1.

But the only values of Tmp1 I see are 1 (when not armed ) and 4( when armed).

"Mavlink encapsulation + SPORT (inav)" mode with inav-specific flight mode encoding to Tmp1 would be nice to have.

macrokernel, 2021-04-02:
Having the same issue with Ardupilot. Would be nice to have this fixed.

ninja-zx11, 2021-05-05:
Yes please fix it!!  Really great system but missing flight modes on ardupilot.

TomekFrs, 2021-06-29:
Working links:
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md
https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/mavlink.c